scholi/pySPM #1: Support for Bruker version 0x09300201

ezatterin, 2018-04-26:
Hi, 

Thanks for the code first of all - very nice! I can read without any issues Bruker files version `0x09010300`, but get weird images when opening newer ones of version `0x09300201` -- in the same way I would if I were to open them with say an old Gwyddion release. Any clue what the difference between them is? If you give me some info I could try to add support for the newer version myself.

Cheers!

scholi, 2018-04-26:
Hi Edoardo,
I'm not aware of the new fileformat. Anyway all the headers are human readable in the old version. If it's still the case, it will be easy to adapt it to the new one.

If it's not human readable you can send me a measurement file with the new version with information about the size (real + pixel sizes) and I can try to reverse engineer the file as I did with the other fileformats of this library.

ezatterin, 2018-04-26:
Thanks for the reply! Nice - the library you linked seems to correctly decode the new version, however it seems that it only looks for specific image channels ('Amplitude', 'Height', etc.) and that can't be changed without tweaking the module directly. I need to go through the code as I need to read other channels.

In the meantime, if you're interested: yes, the header is human-readable. Here's a file: (fyi, these are ferroelectric domains)

[e17087.0_00015.zip](https://github.com/scholi/pySPM/files/1951988/e17087.0_00015.zip)

If it's a quick fix for you I'd be grateful :grin:

scholi, 2018-05-06:
Hi
Sorry I have tested it only now. It actually works fine. The problems come that Bruker sometime record only the backward channel and my library uses by default the forward channel.

You should just use and it will work:

```
A = pySPM.Bruker("e17087.0_00015.spm")
A.list_channels()
A.get_channel(backward=True).show()
```

scholi, 2018-05-06:
Ok I have noticed, that it works, but the value we got are garbaged.
I have fix the library. Please download the last version and test it with your data

ezatterin, 2018-05-08:
Beautiful, it works perfectly. Thanks a million! Out of curiosity, was it just an encoding problem?

closed 2018-05-08 by ezatterin