scholi/pySPM #3: TIFF image meta information

wajahat57, 2018-07-20:
Hi Scholi
Thanks for sharing this toolbox.
Can it be used to extract metadata from an SPM tiff image file such as pixel to world distance mapping  etc?

scholi, 2018-08-11:
Hi,
Sorry for the late reply, I was on holidays...

No I never needed it. If you want to do it, I can push it to the library.

regards

scholi

closed 2018-08-11 by scholi

scholi, 2018-08-14:
Here is a useful link: https://stackoverflow.com/questions/46477712/reading-tiff-image-metadata-in-python

so you can read the data and size information with the information above and create a custom I = SPM_image(data, real=dict(x=real_width,y=real_height,unit=unit_used_for_WidthHeight))

wajahat57, 2018-08-17:
Hi Scholi
Thanks a lot for your response and the link. It was helpful.
What are the real_width and height and the unit?
Are they the real world size of the image field of view in distance metrics such as cm, mm etc?

scholi, 2018-08-17:
yes exactly. Imagine you have an image of 512×512 pixels but it correspond to a real size of 20×20 μm, then x=20, y=20, unit="um" or x=2e-5,y=2e-5,unit="m" for e.g.

wajahat57, 2018-08-20:
[SPM_sample)image.zip](https://github.com/scholi/pySPM/files/2303355/SPM_sample.image.zip)

scholi, 2018-08-21:
TIFF is NO SPM standard. It is basically a non compressed way of storing image which also work with 16bits per pixels making it a possible choice for SPM data.

The problem is that TIFF is not meant to store real size data. Maybe some manufacturer can add the information in the form of tags, but this does not seems to be the case for your image.

Do you know what is the real_size of your image you provided?

scholi, 2018-08-21:
Note: Most manufacturer produce a second file with the same name and another extention woth the metadata. Do you have something like that?

wajahat57, 2018-08-21:
Don't have any such file. Will double check with those involved in image acquisition.

scholi, 2018-08-21:
I guess that those Tif are exported with WinSPM. Which instrument are you using to record the images?

wajahat57, 2018-08-21:
There is an SPM file produced with the following info:
FileName                X       pix   Z      V     I      STS   Date          Time  Comment
110618a1_15.TIF    80.0  512   1.5  0.30  0.05  N     28/6/18   22:43  110618a16

wajahat57, 2018-08-21:
And any idea what V, I and STS are in the information I shared above?

wajahat57, 2018-08-21:
The instrument is Scanning Tunnel Microscope

scholi, 2018-09-28:
just noticed that I forgot to answer.
So for clarity: STM = Scanning Tunneling Microscope, STS=Scanning Tunneling Spectroscopy where you record the current I in function of the voltage V (so the tip-sample bias).

If you still have issue, please tell the people who performed the mesurement to contact me.

kind regards

wajahat57, 2018-09-28:
Ok. Thanks. I will.