scholi/pySPM #5: Normalisation of images

mw00847, 2019-03-05:
Hi Scholi,

Is it possible to normalise images, either by total count or by another image?

Thanks 
Mat

scholi, 2019-03-05:
I don't understand the point to normalize an image. The colorscale is automatically stretch to fit the min/max value, so normalizing the values won't change the color that you see. This can be tuned by using the vmax argument in order to specify the upper range of the colormap which is useful to compare several images. So as an example if you have an image A and B you can show them as follow:

```python
max_ = max(np.max(A.pixels), np.max(B.pixels))
A.show(vmax=max_)
B.show(vmax=max_)
````

mw00847, 2019-03-05:
Sorry I usually normalise images by total to remove any topography issues with the sample.

Thanks I will try your recommendation

scholi, 2019-03-06:
For your info, you can retrieve the total number of count of the ITA object X as follow:

```python
X.get_value("Analysis.SICount")['float']
```

closed 2019-03-27 by scholi