scholi/pySPM #6: Get Data

Groomer67, 2019-04-24:
Hi scholi,

I'm working through your introduction notebook right now. I just got an error 404 when trying to upload your SPM data package, either from the webpage or using 

pip install -U pySPM_data-0.2.4-py3-none-any.whl

Do you have any suggestions?

Ryan

scholi, 2019-04-24:
Hi Ryan, 
you should download it before running pip. you should click on the blue link from the documentation.
the link is https://sftpweb.empa.ch/main.html?download&weblink=f5cddf602b6e8b20325b49a9ec61c48c&realfilename=pySPM_data-0.2.4-py3-none-any.whl

I'll check tomorrow if the link still works. otherwise I'll fix it this week.

regards

Olivier

scholi, 2019-04-24:
I have fixed the link. It should work now.

Enjoy!

closed 2019-04-24 by scholi

Groomer67, 2019-05-03:
Hi Olivier, 
Sorry for the slow reply, but thanks for the help, the fix worked!

dribrahimsalama, 2019-07-02:
Hi scholi,

the link  https://sftpweb.empa.ch/main.html?download&weblink=f5cddf602b6e8b20325b49a9ec61c48c&realfilename=pySPM_data-0.2.4-py3-none-any.whl  is not working again.
would it be possible to upload again, please? 
Ibrahim

scholi, 2019-07-02:
I don't have access to this server anymore. Please use the link in the [README ](https://github.com/scholi/pySPM) at the very bottom.

dribrahimsalama, 2019-07-02:
Thanks scholi