scholi/pySPM #7: Weird topography values

ezatterin, 2019-05-10:
Hi,

The 'Height Sensor' or 'Z sensor' channel values (i.e. the topography), seem to be a bit weird. For example:

```python
scan = PfmFile(filelist[0])

chan = scan.get_channel()
xsize, ysize, unit = scan.get_size() # defined by me

fig, ax = plt.subplots(1,1)

im = ax.imshow(chan.pixels, extent=[0,xsize,0,ysize])
ax.set_title('{0} [{1}]'.format(chan.channel, chan.zscale))

append_colorbar(ax, im, label_size=10) # defined by me
```

Results in, as compared to Gwyddion:

pySPM    |  Gwyddion
:---------:|:--------------:
[image] | [image]

Note that in both cases the minimum of the data has been shifted to zero (but not doing it still shows weird values). The 'nm' label has been produced via `chan.zscale` so should be consistent. Any ideas?

Cheers,

Edoardo

scholi, 2019-05-10:
Thank you for reporting this bug. Something is definitely strange. Could you share your data so that I can test and debug the problem?

Cheers,

Olivier

ezatterin, 2019-05-10:
There you go, thanks!

[e17089.0_00001.spm.zip](https://github.com/scholi/pySPM/files/3165917/e17089.0_00001.spm.zip)

scholi, 2019-05-10:
Ok it should be fixed.  I get:
[image]
which is the correct range, but with an offset as I get negative values

ezatterin, 2019-05-10:
Awesome:

```python
scan = pySPM.Bruker(filelist[0])
chan = scan.get_channel()
p = chan.zero_min().show()
plt.colorbar(p)
````
[image]

Thank you!

scholi, 2019-05-10:
ahah. You know some function better than myself now. I forgot that _zero_min()_ existed...

Keep me updated if you fiund other bugs. I don't use the Bruker module so much anymore and only with the old fileformat...

Regards

rejmar16, 2020-11-26:
Hi,

I am dealing with the same problem. That topography shows weird value.

`import pySPM

ScanB = pySPM.Bruker("SiC_E9_toughness_P1_1.spm")
topoB = ScanB.get_channel('Height')

fig, ax = plt.subplots(1, 2, figsize=(21, 8))
im = topoB.show(ax=ax[0],cmap='hot')
divider1 = make_axes_locatable(ax[0])
cax1 = divider1.append_axes("right", size="5%", pad=0.05, aspect=16)
cbar1 = plt.colorbar(im, cax=cax1)#,  

topoB.plot_profile(0,290,511,290,ax=ax[1],img=ax[0])

plt.tight_layout()
plt.show()`

[image]

Here you can find raw data:

[SiC_E9_toughness_P1_0.zip](https://github.com/scholi/pySPM/files/5599938/SiC_E9_toughness_P1_0.zip)

I am using version 0.2.21.

Any idea, how to fix it.

Thanks,
Martin

scholi, 2021-01-23:
Can you try to execute `scan.get_channel(debug=True)` ? It will then print some scaling information...

scholi, 2021-01-23:
OK I think I found the bug. Somehow Bruker set incorrectly the number of byte per pixel. I could fix it for one of your image. Could you update pySPM and double check with your other images to see if this works?

rejmar16, 2021-01-24:
Hi Scholi,

Thank you for looking at this issue. I upgraded the pySPM library (pip install -U pySPM). However, I do not see any difference. Still version 0.2.21, and I get the same value as before for height.

This `scan.get_channel(debug=True)` returned 'Channel Height Sensor not found'

Martin

scholi, 2021-01-24:
Ah yes I have pushed it to github only. I need to make a new package, but it require testing first. for that you need to download the library from github and execute setup.py as follow
`python setup.py install`

for the bug with the height sensor it is because you should use
`scan.get_channel('Height', debug=True)`

scholi, 2021-01-24:
I have uploaded packaghe 0.2.22. You can get it with pip now:
`pip install -U pySPM`

rejmar16, 2021-01-24:
Thanks, it works for all checked maps (pySPM vs Gwyddion).

Best
Martin

closed 2021-01-24 by scholi