scholi/pySPM #8: SPM_image.get_extent() fails for a Bruker AFM image

ezatterin, 2019-05-10:
The following:

```python
scan = pySPM.Bruker(filename)
chan = scan.get_channel()
chan.get_extent()
```

Results in an error:

```
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-365-32333116e5f9> in <module>
----> 1 chan.get_extent()

c:\python37\lib\site-packages\pySPM\SPM.py in get_extent(self)
    464         Get the image extent in real data
    465         """
--> 466         W = self.size['recorded']['real']['x']
    467         H = self.size['recorded']['real']['y']
    468         return (0, W, 0, H)

KeyError: 'recorded'
```

scholi, 2019-05-10:
This should be fixed now. Can you test it?

ezatterin, 2019-05-10:
Different error:

```
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-37-32333116e5f9> in <module>
----> 1 chan.get_extent()

c:\python37\lib\site-packages\pySPM\SPM.py in get_extent(self)
    464 
    465     def inv_calc_flat(self, d, l=0.1):
--> 466         """
    467         Function used for inverse MFM calculation (inspired from http://qmfm.empa.ch/qmfm/)
    468         The function is in its early devlopment stage as not used by the developed.

KeyError: 'recorded'
```

scholi, 2019-05-10:
This works for me
```python
scan = pySPM.Bruker(os.path.expanduser("~/Dropbox/e17089.0_00001.spm"))
chan = scan.get_channel()
print(chan.get_extent())
chan.pixels -= np.min(chan.pixels)
fig, ax = plt.subplots(1,1)
p = chan.show();
plt.colorbar(p);
```

(0, 500.0, 0, 500.0)
[image]

ezatterin, 2019-05-10:
My bad, I messed up with the pySPM versions. Thanks, closing!

closed 2019-05-10 by ezatterin