scholi/pySPM #9: struct.error: unpack requires a buffer of 80 bytes

dribrahimsalama, 2019-07-02:
Hi Scholi,

I have been trying to use spectra on my data (TOFion) and I received the following error 

**To Reproduce**
Snippet of code creating the error
[image]

 - OS:  windows
- Python 3.7.3 and pySPM version.

Please run the following and attach the result to your issue
`Python 3.7.3 (v3.7.3:ef4ec6ed12, Mar 25 2019, 22:22:05) [MSC v.1916 64 bit (AMD64)]
pySPM 0.2.21
numpy 1.16.4
scipy 1.3.0
matplotlib 3.1.0
```

Thanks
Ibrahim

scholi, 2019-07-02:
Somehow I cannot see your whole code. But the following work for me:

```python
from pySPM_data import get_data
import pySPM.tools.spectra
import pySPM

filename = get_data("Cysteine_B3p_p_01_0.ita")
pySPM.tools.spectra.main(filename)
```

Can you attach you code as code and not as image? Please use the icon "<>" on the top of the editor.

scholi, 2019-07-02:
Now it can be hat the ita file youhave is corrupted. Please run the following:

```python
from pySPM_data import get_data
import hashlib

filename = get_data("Cysteine_B3p_p_01_0.ita")
 
sha256_hash = hashlib.sha256()
with open(filename, "rb") as f:
    for byte_block in iter(lambda: f.read(4096),b""):
        sha256_hash.update(byte_block)
    print(sha256_hash.hexdigest())
```

you should get: 

> 8525c01a563420fe5a97e00655ec57c757e86bb1ac764e89b0565eab0feb52f4

>

scholi, 2019-07-05:
So. What is the status? Can I close the issue?

closed 2019-08-09 by scholi