scholi/pySPM #10: OperationalError: unable to open database file

davislaundon, 2019-07-18:
Hi Olivier, 

I am having trouble loading pySPM into Jupyter Notebook. I have installed successfully pySPM but when i try to 'import pySPM' i get the following error:

OperationalError: unable to open database file

I assume this is something to do with sqlite but, as a python novice, have been unable to troubleshoot it successfully - is there anything I can try? I have all of the prerequisites installed, as well as sqlite3. Apologies if this is issue is trivial. 

Many thanks,

Davis

scholi, 2019-07-31:
Hi Davis,
Sorry for the late reply. I have the same problem on a new machine. I'll try to find some time this week or next one to fix it.

Kind regards

Olivier

scholi, 2019-07-31:
It seems that the database is not copied by pip while installing it... 

You can solve the problem in the mean time to unistall the pip package and then install it manually from github. So download pySPM: [https://github.com/scholi/pySPM/archive/master.zip](https://github.com/scholi/pySPM/archive/master.zip) and unzip it. Then from the folder you downloaded (where setup.py is) run a terminal and type:

```bash
pip uninstall pySPM
python setup.py install
```

Then it should work

scholi, 2019-07-31:
Ok it should be fixed. Please uninstall and reinstall pySPM

```bash
pip uninstall pySPM
pip install pySPM
```

Let me know if this works.

davislaundon, 2019-07-31:
Thanks a lot for your help Olivier - I'll give it a try!

Best

Davis

closed 2019-08-09 by scholi