scholi/pySPM #13: https://github.com/scholi/pySPM_data.  This URL could not be found .

xx2294, 2019-12-03:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Snippet of code creating the error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Information:**
 - OS: [e.g. windows]
- Python and pySPM version.

Please run the following and attach the result to your issue
```python
import sys
import pySPM
import numpy as np
import scipy
import matplotlib as mpl
print("Python",sys.version)
print("pySPM",pySPM.__version__)
print("numpy",np.__version__)
print("scipy",scipy.__version__)
print("matplotlib", mpl.__version__)
```

**Additional context**
Add any other context about the problem here.

scholi, 2019-12-03:
What is the bug? It's normal that this url does not exists.

xx2294, 2019-12-03:
> What is the bug? It's normal that this url does not exists.
> 
> Where did you found this link?

Hello! I found this link in the Document. I want to download the pySPM_data library which provides all example data files, but according to your link (https://github.com/scholi/pySPM_data) it can't be open.  It seems that the information has been deleted?

xx2294, 2019-12-03:
Could you provide  the pySPM_data library for me? Thank you very much!

scholi, 2019-12-03:
Ah yes right. Thank you for spotting it. I'll try to fix it.
In the meantime just read the Readme till the very end and you will discover the proper link.

xx2294, 2019-12-03:
Looking forward to your link! Thank you for your help again!

closed 2019-12-03 by scholi

scholi, 2019-12-04:
The link for pySPM_data is in the readme. You even made a nice screenshot of the link ;)

xx2294, 2019-12-05:
I know the link for pySPM_data is in the readme, but the link is invaild.
[image]