scholi/pySPM #14: D:\Anaconda\lib\site-packages\pySPM\data\CristA_0002.xml

xx2294, 2019-12-04:
Sorry to bother you again！
[image]
In the 'data' directory, I couldn't find the file called CristA_0002.xml.
Thank you!

scholi, 2019-12-04:
It is not supposed to be there. The data are part of the **pySPM_data** package, not pySPM as you show in your screenshot. There is also a python function to load them, but for that you need to install them and not have them laying around in your download folder as in your screenshot.

xx2294, 2019-12-04:
Where can I download pySPM_data package for subsequent installations?
Thank you for yor answer!

scholi, 2019-12-04:
Question already answered see #13

closed 2019-12-04 by scholi