scholi/pySPM #16: Wired z axis values

Thoronfin, 2021-01-14:
Hi,

the z scale (height) values from my code seems a bit wired.

```
filename="CaptureFile.0_00019.spm"
ScanB = pySPM.Bruker(filename)
Height = ScanB.get_channel("Height")
Height2 = Height.correct_plane(inline=False)
fig, ax = plt.subplots(1,1, figsize=(5,5))
p=Height2.zero_min().show(ax=ax, cmap="inferno", title="")
cbar=plt.colorbar(p,fraction=0.046)
cbar.set_label(label=Height2.zscale,rotation=270, labelpad=10)
plt.tight_layout()
```

This is the result from the code:
[image]

This is the same picture opened through Gwyddion:
[image]

This is the source file:
[CaptureFile.0_00019.zip](https://github.com/scholi/pySPM/files/5815013/CaptureFile.0_00019.zip)

I plotted the same graphs using a file from an other issue (see below). In that file there is a

> [ZSensor] "Height Sensor" 

and the python plot is identical to the Gwyddion graph.

[e17089.0_00001.zip](https://github.com/scholi/pySPM/files/5815064/e17089.0_00001.zip)

Cheers

scholi, 2021-01-23:
Hi,
Yes there is some issue with the Bruker files.Bruker changed there file format with time and have some encoding issue. I don't have access to an instrument, so I cannot do any testing. If someone can find and correct the problem he can send a push request and I'll merge it to the library

scholi, 2021-01-23:
This issue a a duplicata of https://github.com/scholi/pySPM/issues/7 ⇒ closing this one

closed 2021-01-23 by scholi

scholi, 2021-01-23:
I have fixed the library. Please update and test
[image]

Thoronfin, 2021-01-25:
Thanks, it seems work