scholi/pySPM #17: Issue with median of differences correction

Thoronfin, 2021-01-29:
Hi,

I noticed a difference in the median difference correction between pySPM and Gwyddion.
The first row shows the each step of the image correction in python, the second one shows the corresponding Gwyddion images. Only the last figures are different even if the are both correceted by the median of differences.

[image]

Here you can find the source code:

```
import numpy as np
import matplotlib.pyplot as plt
import matplotlib as mpl
import pySPM

file="01201542.0_00001"
filename=file+".spm"

ScanB = pySPM.Bruker(filename)
Height = ScanB.get_channel("Height")
Height_corrected = Height.correct_plane(inline=False)

import copy

Height_corrected_1 = copy.deepcopy(Height_corrected) # make deepcopy of object otherwise you will just change the original
Height_corrected_1.correct_median_diff(inline=True)
fig, ax = plt.subplots(1,3,figsize=(15,7))
Height.show(ax=ax[0], title="No corrections", cmap="viridis")
Height_corrected.show(ax=ax[1], title="Plane correction", cmap="viridis");
Height_corrected_1.show(ax=ax[2], title="Plane correction + median of differences", cmap="viridis");
plt.tight_layout()
plt.savefig(file+"comparison.png")
```

Here you can find the source file:
[01201542.0_00001.zip](https://github.com/scholi/pySPM/files/5893679/01201542.0_00001.zip)

I'm currently using juyter notebook on windows 10.
These are the versions of the libraries:

```
Python 3.8.5 (default, Sep  3 2020, 21:29:08) [MSC v.1916 64 bit (AMD64)]
pySPM 0.2.22
numpy 1.19.2
scipy 1.5.2
matplotlib 3.3.2
```

Cheers

scholi, 2021-01-29:
There was a typo in the code. Thank you for spotting it. Please update and try again

Thoronfin, 2021-01-29:
I checked it and it works properly now

Thank you

closed 2021-01-29 by scholi