scholi/pySPM #18: "li" variable not defined in ITM.py

jablevin, 2021-07-14:
When running >>> res = AI.get_raw_spectrum(ROI=[ROI1, ROI2], prog=True, sf=sf, k0=k0) 
I get the error that local variable "li" is referenced before assignment. 

The code from the file is below. The error arises near the end "for k in li"

elif type(ROI) in [list, tuple]:
            multi_roi = True
            Spectrum = np.zeros((number_channels, len(ROI)), dtype=np.float32)
            for s in T:
                raw = self.get_raw_raw_data(s)
                rawv = struct.unpack('<{}I'.format(len(raw)//4), raw)
                i = 0
                while i < len(rawv):
                    b = rawv[i]
                    if b & 0xc000000000:
                        x = b & 0x0fffffff
                        y = rawv[i+1] & 0x0fffffff
                        dt = dts[x]
                        fp = dt%1
                        ip = int(dt)
                        li = []
                        for k, R in enumerate(ROI):
                            if R[y,x]:
                                li.append(k)
                        i += 3
                    else:
                        for k in li:
                                Spectrum[b-ip, k] += (1-fp)
                                Spectrum[b-ip-1, k] += fp
                        i += 1

scholi, 2021-07-14:
There is definitely a python bug, but I'm not sure to remember correctly how the rawSpectrum is encoded to correct it and beeing sure that the data you get make sense....
I guess  you should predefine li before the while. So add
````python
li=[]
````  at line 1050

scholi, 2021-08-04:
Can I get some feedback? Does this solution works?

andrealucero, 2021-08-21:
Hello. I have a similar error when I try to run the following cell from the "Introduction to pySPM a python library for ToF-SIMS data analysis":
AI = pySPM.ITM(get_data("Cysteine_B3p_p_01.itm"))
sf, k0 = A.auto_mass_cal()
try:
    # Reconstruction is a slow porcess, so the data are stored in order to speed up the running speed of this notebook
    res = pySPM.utils.load("Intro_pySPM_tofsims", 'Cysteine_ROI')
except:
    res = AI.get_raw_spectrum(ROI=[ROI1, ROI2], prog=True, sf=sf, k0=k0)
    pySPM.utils.save("Intro_pySPM_tofsims", Cysteine_ROI=res)

The error is:
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-43-9d019fae432e> in <module>
      4     # Reconstruction is a slow porcess, so the data are stored in order to speed up the running speed of this notebook
----> 5     res = pySPM.utils.load("Intro_pySPM_tofsims", 'Cysteine_ROI')
      6 except:

~\Anaconda3\lib\site-packages\pySPM\utils\save.py in load(filename, *keys)
     92     """
---> 93     filename = findPKZ(filename)
     94     if len(keys) == 0:

~\Anaconda3\lib\site-packages\pySPM\utils\save.py in findPKZ(filename)
     36     if not os.path.exists(filename):
---> 37         raise IOError("File \"{}\" not found".format(filename))
     38     return filename

OSError: File "Intro_pySPM_tofsims.pkz" not found

During handling of the above exception, another exception occurred:

UnboundLocalError                         Traceback (most recent call last)
<ipython-input-43-9d019fae432e> in <module>
      5     res = pySPM.utils.load("Intro_pySPM_tofsims", 'Cysteine_ROI')
      6 except:
----> 7     res = AI.get_raw_spectrum(ROI=[ROI1, ROI2], prog=True, sf=sf, k0=k0)
      8     pySPM.utils.save("Intro_pySPM_tofsims", Cysteine_ROI=res)

~\Anaconda3\lib\site-packages\pySPM\ITM.py in get_raw_spectrum(self, scans, ROI, FOVcorr, deadTimeCorr, **kargs)
   1063                                 li.append(k)
   1064                         i += 3
-> 1065                     else:
   1066                         for k in li:
   1067                                 Spectrum[b-ip, k] += (1-fp)

UnboundLocalError: local variable 'li' referenced before assignment

Looks like it's not only the assignment of "li". I'm not an expert, so I would appreciate your help.
Thank you in advance.

scholi, 2021-08-30:
and? did you try the proposed solution? Does it work?

andrealucero, 2021-08-30:
Yes. I did what you suggested before and it worked. Thank you!

scholi, 2021-08-30:
OK. I'll add it for next release then. Thank you for the feedback.

closed 2021-08-30 by scholi