scholi/pySPM #19: reuse file format reader in afmformats

paulmueller, 2021-08-03:
Hi,

I am the author of [afmformats](https://afmformats.readthedocs.io) (used by [PyJibe](https://pyjibe.readthedocs.io/)) and I was wondering whether it would make sense to incorporate the file-reader functionalities of pySPM in afmformats.

1. I have previously only worked with force-distance data. Are the file formats you have used here plain image files or do they contain the raw force-distance trace data (i.e. cantilever deflection vs. piezo position)?
2. What do you think about this? Does it even make sense?
3. My initial thought was to just make pySPM a dependency of afmformats, but then I saw that pySPM depends on several other packages which I would like to avoid. I see that pySPM is licensed under the Apache 2.0 license which would be compatible with the MIT license (afmformats), as far as I know.

Cross-ref: https://github.com/AFM-analysis/afmformats/issues/18

scholi, 2021-08-04:
Hi Paul,
The data contained in the file will always depend on the experiment you perform. Usually they are 2D arrays or 1D arrays and not multidimentional array such as a force-distance per pixel. From what I see in your package afmformats you just get 1D force-distance curve. Is that correct?

Which new fileformat do you want to read? It makes no sense to implement new file format just to add more fileformat if you don't use them and cannot debug them properly.

paulmueller, 2021-08-04:
Yes, afmformats supports 1D force-distance curves, but can handle quantitative map data (force curves aranged on a grid).
I am currently just taking a look around to expand general data file format support.

scholi, 2021-08-04:
In anycase pySPM is under Apache license and you are free to use it as a dependency or to reuse any python script. Just keep the list of all contributing authors in the header.

closed 2021-08-04 by scholi