scholi/pySPM #20: Defining regions of interest (ROI)

andrealucero, 2021-08-30:
Hi! I've been reading (and running) the "Introduction to pySPM a python library for ToF-SIMS data analysis" tutorial. When it comes to defining a ROI, the example you provide is by thresholding the SI data. I was wondering if there are other options to define a ROI, for example: drawing a rectangle in the SI intensity image (and then obtain the spectrum from the area inside the rectangle).
Thanks again!

scholi, 2021-08-30:
did you try the help file?
help(pySPM.ITM.get_raw_spectrum)
You should get:
 ROI: Region Of Interest. It can be:
            1) None: All pixels are used
            2) A list of images. The pixels will be added to the spectrum i if the value of the i-th image is True for that pixel
            3) An image with integer value. The current pixels will be added to the i-th spectrum if the value of ROI at that pixel is i.

Then you can create a ROI with a one values in a rectangular portion.

andrealucero, 2021-08-31:
I used pySPM.utils.plotMask with a mask I created with the rectangle to have a proper image of the roi and then used pySPM.ITM.get_raw_spectrum (with the same mask). Looks like it works :)

closed 2021-08-31 by scholi