scholi/pySPM #22: Nonsquare scans Bruker files (.spm)

GGGGGuillaume, 2021-11-22:
Hello, thank you very much for your work, it is great (and very time efficient) to reopen the Bruker files with python ! 

I have a little trouble with opening scans that are non square (where the aspect ratio is not 1:1).  For example, I have a 2:1 aspect ratio scan, X size is 10 µm and Y size = X size/2 = 5 µm, 256 pixels in X, 128 pixels in Y. When I use pySPM, the scan becomes a 10*10 µm scan...

Thanks in advance,

Guillaume

scholi, 2021-11-22:
Can you send me some data to test it?

GGGGGuillaume, 2021-11-23:
[scan_test.zip](https://github.com/scholi/pySPM/files/7586450/scan_test.zip)

Of course, cf. attached file !

scholi, 2021-11-24:
I found that Bruker is using an Aspect Ratio info and could fix it so that it works for the example you have sent me.
Unfortunately I don't have access to data anymore. Could you pull the git (not pip) and make some tests to see if the results makes sense?

scholi, 2021-11-24:
```python
from pySPM import Bruker
import os.path

b = Bruker(os.path.expanduser("~/desktop/scan_test.spm"))
c = b.get_channel("Height", debug=True)
c.show()
```
[image]

GGGGGuillaume, 2021-11-25:
I have tested it with a few scans (different aspect ratios), it works perfectly ! Thanks you very much for your quick assistance (and thanks again for the library, it will be very useful for me).
[image]

closed 2021-11-25 by GGGGGuillaume