scholi/pySPM #26: Error when opening an ITM File generated by Surface Lab 7

teolombardo, 2022-05-25:
Dear developer(s),

I was trying to use pySPM for reading ITM files generated by SurfaceLab 7.X. This is the version that creates the .itax files instead of the .ita, but, together with it, the software also prints .itm files, which I hoped could be read by pySPM. Is it incorrect?

In particular, I tried something very simple:

import pySPM
AI = pySPM.ITM("test.itm", readonly=True)

But it raises the following error (if I try removing the readonly option, of course nothing change) :
---------------------------------------------------------------------------
MissingBlock                              Traceback (most recent call last)
Input In [3], in <cell line: 6>()
      2 path_test = path + "\\2022_03_24_non_coated_NMC_spectrometry_positive_01_TL.itm"
      4 print(path_test)
----> 6 AI = pySPM.ITM("test.itm", readonly=True)

File ~\AppData\Local\Programs\Python\Python310\lib\site-packages\pySPM\ITM.py:114, in ITM.__init__(self, filename, debug, readonly, precond, label)
    112     except:
    113         self.Nscan = None
--> 114 self.spp = self.root.goto("propend/Registration.Raster.ShotsPerPixel").get_key_value()['int']
    115 try:
    116     R = [z for z in self.root.goto('MassIntervalList').get_list() if z['name'] == 'mi']

File ~\AppData\Local\Programs\Python\Python310\lib\site-packages\pySPM\Block.py:545, in Block.goto(self, path, lazy)
    543         p = e['name']
    544         idx = e['id']
--> 545     s = s.goto_item(p, idx, lazy=lazy)
    546 return s

File ~\AppData\Local\Programs\Python\Python310\lib\site-packages\pySPM\Block.py:488, in Block.goto_item(self, name, idx, lazy)
    482 @deprecated("gotoItem")
    483 def goto_item(self, name, idx=0, lazy=False):
    484     """
    485     Return a new Block instance of a child of the current Block
    486     name: name of the children's block
    487     """
--> 488     Idx = self.get_index(name, idx, lazy=lazy)
    489     self.f.seek(Idx)
    490     return Block(self.f, parent=self)

File ~\AppData\Local\Programs\Python\Python310\lib\site-packages\pySPM\Block.py:515, in Block.get_index(self, name, idx, lazy)
    513             return l['bidx']
    514         i+=1
--> 515 raise MissingBlock(self, name, idx)

MissingBlock: Missing block "/propend/Registration.Raster.ShotsPerPixel" with index 0

Thanks in advance for your help, I am really looking forward for using this library!

Teo Lombardo

scholi, 2022-05-29:
Duplicate of: https://github.com/scholi/pySPM/issues/23

scholi, 2022-05-29:
ITM files are huge, but if you find a way to share it with me I can try to see if I can help...

closed 2023-07-11 by dineshpinto