scholi/pySPM #30: Unable to load an approach curve from Bruker Icon

Valentinfonck, 2023-01-02:
Hi scholi,

Thank you for your very handy library. I have been trying to use it on approach-retract curve for AFM and I've been running into troubles. Is this feature already implemented ?

**List of channels empty when importing an approach curve**
While the creation of the object works, the library writes down an empty list of channels when loading the spm file of an approach curve of a Bruker Icon Dimension (using the ramp option in the dedicated software.

`        meas = Bruker(path)
        meas.list_channels()
        data = meas.get_channel()`

**Expected behavior**
I'm trying to recover this kind of data
[image]

**Information:**
Windows 10 64bits
Python 3.8.8 (default, Apr 13 2021, 15:08:03) [MSC v.1916 64 bit (AMD64)]
pySPM 0.2.23
numpy 1.20.1
scipy 1.6.2
matplotlib 3.3.4

scholi, 2023-01-07:
Hi,
What is the output of 
```python
meas = Bruker(path)
meas.list_channels()
data = meas.get_channel(channel_name)
```

Wang-stables, 2023-02-22:
Hello,
Also have the same problem, actually I don't know how to open the .spm file with pySPM because the example show on documentation only introduce the .ita file
And for the code below:
meas = pySPM.Bruker("CaptureFile.0_00667.spm")
meas.list_channels()
the output is below:
Channels
========

scholi, 2023-03-05:
.spm is not a Bruker file format, so you cannot use the Bruker module to load it. Unfortunately many instruments are using the spm extension: http://gwyddion.net/documentation/user-guide-en/file-formats.html
Could you tell me which one?

Wang-stables, 2023-03-06:
Thank you for your asking, I'm asking others if they know the true format
is, because I can't sure about this.

Wang-stables, 2023-03-07:
Ok, they tell me that we are using Veeco ll and Veeco lll, also we usually
use the nanoscope analysis 3.00 for analysis.
Again, thank you for sending the email for discussion.

closed 2023-07-27 by dineshpinto