scholi/pySPM #32: Error while using correct_lines/ correct_slope on .spm- (Bruker-) data with different x-,y-dimension

lukas271828, 2023-04-11:
**Describe the bug**
A clear and concise description of what the bug is.

When spm data is loaded while using the Bruker and SPM module, the functions correct_lines and correct_slope do not work and return the following Error: 

"self.pixels -= np.tile(np.mean(self.pixels, axis=1).T, (self.pixels.shape[0], 1)).T
ValueError: operands could not be broadcast together with shapes (128,1024) (128,128) (128,1024)"

Furthermore, the plot with .show(pixels='False') returns an image with wrong y-dimension: here both axes are shown with the same range of the x axes.

The function works fine if i use an image with same x and y dimension.

**To Reproduce**
Snippet of code creating the error: 

filename = 'testfile.spm'
scan = pySPM.Bruker(filename)
topoB.correct_lines(inline=True)
topoB.show()

topoB = scan.get_channel("Height Sensor")

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Information:**
 - OS: [e.g. windows]
- Python and pySPM version.

Please run the following and attach the result to your issue
```python
import sys
import pySPM
import numpy as np
import scipy
import matplotlib as mpl
print("Python",sys.version)
print("pySPM",pySPM.__version__)
print("numpy",np.__version__)
print("scipy",scipy.__version__)
print("matplotlib", mpl.__version__)
```
Python 3.9.13 (main, Aug 25 2022, 23:51:50) [MSC v.1916 64 bit (AMD64)]
pySPM 0.2.23
numpy 1.21.5
scipy 1.9.1
matplotlib 3.5.2

**Additional context**
Add any other context about the problem here.

dineshpinto, 2023-04-11:
So this is a known limitation, where the `correct_` functions only work on square arrays. I don't have some raw data to test with atm, but can you try this and see if it works correctly

```
import numpy as np
import pySPM

filename = 'testfile.spm'
scan = pySPM.Bruker(filename)
topoB = scan.get_channel("Height Sensor")

# the correct lines logic
topoB.pixels -= np.tile(np.mean(topoB.pixels, axis=1).T, (topoB.pixels.shape[1], 1)).T

# check if the result is correct
topoB.show(pixels=True)
```

lukas271828, 2023-04-12:
Hello,
your suggestion worked perfectly and solves the problem for me. 
Thank you for the solution and the qucik reply!

closed 2023-04-12 by dineshpinto