scholi/pySPM #34: Unable to read a SXM file from Nanonis

Codissimo112, 2023-07-22:
**Describe the bug**
I tried in the last days to use your package for read .sxm data to python to get an array of the Topo signal of my measurements. By doing that I used your documentation and installed the both .whl datas you need to run your code. I simply tried to copy your section bout reading .sxm file and get always the error:

> File c:\users\rbrue\nextcloud\python projekte\read nanonis.py:24
>     S = pySPM.SXM(filename)
> 
> File ~\AppData\Roaming\Python\Python311\site-packages\pySPM\SXM.py:36 in __init__
>     self.header[key].append(l.decode('ascii').split())
> UnicodeDecodeError: 'ascii' codec can't decode byte 0xb0 in position 11: ordinal not in range(128)

**To Reproduce**
```
#import nanonis_load
import numpy as np
import matplotlib.pyplot as plt
import matplotlib as mpl

import os
from IPython import display
import pySPM

from pySPM_data import get_data

filename = get_data('C:\\Users\\rbrue\\Nextcloud\\Data Lab 013\\Sample 09_22\\09_2022\\FeTa110_20220923_CrTip_9935R_4.2K_2.4T_011.sxm')
S = pySPM.SXM(filename)
```

**Expected behavior**
Basically I expect that the reading work properly.

**Information:**

- Windows 10
- Python 3.11.3
- pySPM pyspm-0.3.2-py3-none-any.whl
- pySPM_data pySPM_data-0.2.4-py3-none-any.whl

Please run the following and attach the result to your issue

```Python 3.11.3 (tags/v3.11.3:f3909b8, Apr  4 2023, 23:49:59) [MSC v.1934 64 bit (AMD64)]
numpy 1.25.1
scipy 1.11.1
matplotlib 3.7.2
```

**Additional context**
I comment also the line where the version of pySPM is trying to print because there I get this error:

> AttributeError: module 'pySPM' has no attribute '__version__'

That's also why I don't show the version of pySPM in the last section.
Maybe that problem is linked to the other but I am not sure.

dineshpinto, 2023-07-22:
Hi, I double checked the SXM loader with Nanonis measurements and it works correctly. Try pointing it directly to your measurement data:

```python
from pathlib import Path
import pySPM

p = Path("C:\\Users\\rbrue\\Nextcloud\\Data Lab 013\\Sample 09_22\\09_2022\\FeTa110_20220923_CrTip_9935R_4.2K_2.4T_011.sxm")
afm = pySPM.SXM(p)
afm.list_channels()
```

Codissimo112, 2023-07-22:
Hey,

first of all thanks a lot for this quick response!!
Unfortunately it is giving me exactly the same error meassage like before.

> afm = pySPM.SXM(p)
> Traceback (most recent call last):
> 
>   Cell In[27], line 1
>     afm = pySPM.SXM(p)
> 
>   File ~\AppData\Roaming\Python\Python311\site-packages\pySPM\SXM.py:36 in __init__
>     self.header[key].append(l.decode('ascii').split())
> 
> UnicodeDecodeError: 'ascii' codec can't decode byte 0xb0 in position 11: ordinal not in range(128)

dineshpinto, 2023-07-22:
Hmm, okay. 

This looks like something with the parser itself then. If you send me the SXM file, i can take a look (annual.fallout_0z@icloud.com).

Also what is the Nanonis HW you are using?

Codissimo112, 2023-07-24:
> Hmm, okay.
> 
> This looks like something with the parser itself then. If you send me the SXM file, i can take a look ([annual.fallout_0z@icloud.com](mailto:annual.fallout_0z@icloud.com)).
> 
> Also what is the Nanonis HW you are using?

Did u get my email ?

dineshpinto, 2023-07-24:
Try the file directly output from Nanonis, and see if that works.

dineshpinto, 2023-07-24:
Okay, so the structure of this file is correct. The reason for this error is the unicode characters in the file (the parser assumes ascii). This is fixed in the latest version of pySPM 0.3.3.

Thanks for reporting it.

closed 2023-07-24 by dineshpinto

Codissimo112, 2023-07-24:
So you mean it is fixed because you fixed it now or am I using an old version of pySPM?

Codissimo112, 2023-07-24:
Thanks a lot, it is working!!! Wow I really appreciate your time you spend for solving this issue!!