scholi/pySPM #38: Bruker get_channel floating point parsing error

SylviaWhittle, 2024-01-09:
**Describe the bug**
Hi there 😄 

We have been using pySPM for a while now to open Bruker .spm files, and it has been going seamlessly. Today we have noticed that on files that previously definitely loaded properly, it now throws this error. The files are non-square in their aspect ratio.

The issue appears to be the handling of floating point values in the `get_channel()` method of the `Bruker` class.

What is weird is that this error happens on both version `0.2.23` and `0.5.1` and manually setting the versions to previous versions does not fix it.

**To Reproduce**
file used:
[not_working_file.spm.zip](https://github.com/scholi/pySPM/files/13875221/not_working_file.spm.zip)

I had to compress it since GitHub won't allow `.spm` uploads. I hope this doesn't corrupt it.

```python
from pathlib import Path

import pySPM

print("pySPM version: ", pySPM.__version__)

file = Path('/Users/sylvi/Downloads/not_working_file.spm')

data = pySPM.Bruker(file)

data.list_channels()

height = data.get_channel('Height')
```

**Expected behavior**
Loading the height channel data as usual.

**Screenshots**
[image]

**Information:**

- OS: Windows 10 (intel x86), macOS Sonoma (ARM M1 Pro) 
- Python version: 3.10.9
- PySPM version: 0.5.1

Please run the following and attach the result to your issue

```python
import sys
import pySPM
import numpy as np
import scipy
import matplotlib as mpl
print("Python",sys.version)
print("pySPM",pySPM.__version__)
print("numpy",np.__version__)
print("scipy",scipy.__version__)
print("matplotlib", mpl.__version__)
```

```
Python 3.10.9 | packaged by conda-forge | (main, Feb  2 2023, 20:26:08) [Clang 14.0.6 ]
pySPM 0.5.1
numpy 1.26.3
scipy 1.10.1
matplotlib 3.7.1
```

scholi, 2024-01-09:
It seems that the aspect ratio is a float and not a pair of int.
Is this picture correct or the aspect ratio is the otherway arround?
[image]

SylviaWhittle, 2024-01-09:
Thank you for your attention

B is correct.

I have been told that we have successfully used pySPM on these images before which is strange... It seems that it shouldn't have ever worked if the code assumed a pair of integers.

scholi, 2024-01-10:
What is the real size of your image? 300μm horizontally? but for the vertical axis Gwyddion tells me about 210μm while the aspect ratio of 0.51216 of 300μm is 153μm which is strange...

SylviaWhittle, 2024-01-11:
The real size of the image as reported by Nanoscope, are 300nm x 210nm (0.7 aspect ratio)
[image]

I also held a tape measure up to my screen and confirmed an aspect ratio of 0.7.

We have no idea where the 0.51216 aspect ratio could come from. It is very strange. It is also strange that these files loaded perfectly fine before about December 22nd 2023. Perhaps there is some user error in there somewhere...

Your branch works on both the files that were not previously able to be loaded. I cannot share the second one as it is a collaborator's data and we don't have permission to share it.

Here is a screenshot of it working:
[image]

Thank you for your time, it is very much appreciated by our team 😄

scholi, 2024-01-11:
Ok good to hear that it works :). So I will merge this branch to master.

closed 2024-01-11 by scholi