scholi/pySPM #40: Need Engage X and Y Pos info

nihtmusic, 2024-01-18:
I'm trying to pull out "Engage X Pos" and "Engage Y Pos" from a bruker file.

[image]

I don't see this in the existing list I get from layers:

[image]

Is there another way to get that info from the file?  It seems you are mostly only pulling the Ciao image list...not the File List shown in the first image that is a screen shot of nanoscope. 

Thanks,

Kurt

dineshpinto, 2024-01-20:
You can try and manually parse the information, example snippet:

```python
import pySPM

with open("<path to afm file>", "rb") as f:
    line = f.readline().rstrip().replace(b"\\", b"")
    while line != b"*File list end":
        line = f.readline().rstrip().replace(b"\\", b"")
        if b"Engage X Pos" in line or b"Engage Y Pos" in line:
            print(line)
```

from which you can extract the number and unit.

nihtmusic, 2024-01-23:
Hey!  That worked!  Thank you!

Kurt

closed 2024-01-23 by nihtmusic