scholi/pySPM #45: Example files not available

AlexHenderson, 2024-05-10:
**Describe the bug**
Example files not available

**To Reproduce**
pip install -U pySPM_data-0.2.4-py3-none-any.whl

or

https://sftpweb.empa.ch/main.html?download&weblink=f5cddf602b6e8b20325b49a9ec61c48c&realfilename=pySPM_data-0.2.4-py3-none-any.whl

gives a 404 error

**Files**
NA

**System Information**
>>> print("Platform", sys.platform)
Platform win32
>>> print("Python", sys.version)
Python 3.12.3 (tags/v3.12.3:f6650f9, Apr  9 2024, 14:05:25) [MSC v.1938 64 bit (AMD64)]
>>> print("pySPM", pySPM.__version__)
pySPM 0.6.1

dineshpinto, 2024-05-10:
There is a link to `pySPM_data-0.2.4-py3-none-any.whl` in the [documentation](https://nbviewer.org/github/scholi/pySPM/blob/master/doc/pySPM%20Documentation.ipynb)

`pip install <path>/pySPM_data-0.2.4-py3-none-any.whl`

closed 2024-05-10 by dineshpinto

AlexHenderson, 2024-05-10:
Thanks for pointing that out. 

The link in "Introduction to pySPM for ToF-SIMS data.ipynb" needs to be updated though.