scholi/pySPM #46: add_scale(1e-9) gives wrong scalebar

kabiel, 2024-06-14:
**Describe the bug**
add_scale(1e-9) gives wrong scalebar:
[image]

**To Reproduce**
test = spm.SXM(test_file).get_channel('[P8]_Lockin_X')
test.add_scale(1e-9) # adds scale bar; something wrong
test.show()
plt.show()

**Files**
If applicable, upload or link the file you are having a problem with.

**System Information**
Run the following and attach the result with your issue:

```python
import sys
import pySPM

print("Platform", sys.platform)
print("Python", sys.version)
print("pySPM", pySPM.__version__)
```

**Additional context**
Add any other context about the problem here.

scholi, 2024-06-15:
OK it also took me some time to figure this.
First: you need to plot your image first, then use add_scale
Second: add_scale expect a value (unitless) which is expressed in the same unit as the plot.
Can you try:

```python
test = spm.SXM(test_file).get_channel('[P8]_Lockin_X')
test.show()
test.add_scale(1)
plt.show()
```

kabiel, 2024-06-17:
Great, thank you!! It works now.  Von: Olivier Scholder ***@***.***>Gesendet: Samstag, Juni 15, 2024 2:47 PMAn: scholi/pySPM ***@***.***>Cc: kabiel ***@***.***>; Author ***@***.***>Betreff: Re: [scholi/pySPM] add_scale(1e-9) gives wrong scalebar (Issue #46) 
OK it also took me some time to figure this.
First: you need to plot your image first, then use add_scale
Second: add_scale expect a value (unitless) which is expressed in the same unit as the plot.
Can you try:
test = spm.SXM(test_file).get_channel('[P8]_Lockin_X')
test.show()
test.add_scale(1)
plt.show()